geerlingguy/top500-benchmark #2: Failures building MPI with gfortran-9 on  5.4.0-1071-raspi #81-Ubuntu  (20.04)

brauhausdc, 2022-11-10:
Tried to run this on a cluster of pis running Ubuntu 20.04 - it failed trying to build mpi. 

Is there a reason not to use the OS provided version of mpich, if available? 

Error here: 

`
failed: [wrk-0002] (item=./configure --with-device=ch3:sock FFLAGS=-fallow-argument-mismatch) => changed=true 
  ansible_loop_var: item
  cmd:
  - ./configure
  - --with-device=ch3:sock
  - FFLAGS=-fallow-argument-mismatch
  delta: '0:01:58.104863'
  end: '2022-11-10 14:31:28.537145'
  item: ./configure --with-device=ch3:sock FFLAGS=-fallow-argument-mismatch
  msg: non-zero return code
  rc: 1
  start: '2022-11-10 14:29:30.432282'
  stderr: |-
    configure: WARNING: X11 not found; GL disabled
    configure: WARNING: compilation failed
    configure: error: ****  Incompatible Fortran and C Object File Types!  ****
    F77 Object File Type produced by "gfortran -fallow-argument-mismatch " is : : cannot open `' (No such file or directory).
     C  Object File Type produced by "gcc    -O2" is : : ELF 64-bit LSB relocatable, ARM aarch64, version 1 (SYSV), not stripped.
  stderr_lines: <omitted>
  stdout: |-
... 
`

geerlingguy, 2022-11-10:
@brauhausdc - Honestly I had the same issue on Ubuntu on GitHub Actions, not sure why exactly.

Regarding why I'm compiling mpich, I recall there being some issue with the version that was available for the Pi at the time I initially wrote this playbook for the Turing Pi 2 cluster. That problem may or may not be present anymore, though...

MyloWhylo, 2023-06-09:
I was attempting to run this on WSL, and I managed to solve the issue by installing `gfortran-10` with apt, then changing line 56 in main.yml to `- ./configure --with-device=ch3:sock FFLAGS=-fallow-argument-mismatch FC=gfortran-10`. Hopefully this helps someone else out there!

closed 2023-12-14 by github-actions[bot]