geerlingguy/top500-benchmark #7: Install ATLAS Task failed

Casper042, 2022-11-22:
**hosts.ini**
`# For single node benchmarking (default), use this:
[cluster]
127.0.0.1 ansible_connection=local

# For cluster benchmarking, delete everything above this line and uncomment:
# [cluster]
# node-01.local
# node-02.local
# node-03.local
# node-04.local
# node-05.local
#
# [cluster:vars]
# ansible_user=username`

**config.yml**
`---
# Working directory where HPL and associated applications will be compiled.
hpl_root: /opt/top500

# Home directory of the user for whom SSH keys will be configured.
ssh_user: dantemp
ssh_user_home: /home/dantemp

# Specify manually if needed for mixed-RAM-capacity clusters.
ram_in_gb: "{{ ( ansible_memtotal_mb / 1024 * 0.75 ) | int | abs }}"

# Count the nodes for accurate HPL.dat calculations.
nodecount: "{{ ansible_play_hosts | length | int }}"

# HPL.dat configuration options.
# See: https://www.advancedclustering.com/act_kb/tune-hpl-dat-file/
# See also: https://hpl-calculator.sourceforge.net/HPL-HowTo.pdf
hpl_dat_opts:
  # sqrt((Memory in GB * 1024 * 1024 * 1024 * Node count) / 8) * 0.9
  Ns: "{{ (((((ram_in_gb | int) * 1024 * 1024 * 1024 * (nodecount | int)) / 8) | root) * 0.90) | int }}"
  NBs: 256
  # (P * Q) should be roughly equivalent to total core count, with Qs higher.
  # If running on a single system, Ps should be 1 and Qs should be core count.
  Ps: 1
  Qs: 192`

**ATLAS error:**
`TASK [Install ATLAS (takes a LONG time).] ********************************************************************************************************
failed: [127.0.0.1] (item=../ATLAS/configure) => changed=true
  ansible_loop_var: item
  cmd:
  - ../ATLAS/configure
  delta: '0:00:04.281680'
  end: '2022-11-22 06:54:52.777140'
  item: ../ATLAS/configure
  msg: non-zero return code
  rc: 1
  start: '2022-11-22 06:54:48.495460'
  stderr: |-
    /usr/bin/ld: /tmp/ccMxBJwV.o: in function `ATL_tmpnam':
    /opt/top500/tmp/atlas-build/../ATLAS//CONFIG/include/atlas_sys.h:224: warning: the use of `tmpnam' is dangerous, better use `mkstemp'
    /usr/bin/ld: probe_OS.o: in function `ATL_tmpnam':
    /opt/top500/tmp/atlas-build/../ATLAS//CONFIG/include/atlas_sys.h:224: warning: the use of `tmpnam' is dangerous, better use `mkstemp'
    /usr/bin/ld: probe_asm.o: in function `ATL_tmpnam':
    /opt/top500/tmp/atlas-build/../ATLAS//CONFIG/include/atlas_sys.h:224: warning: the use of `tmpnam' is dangerous, better use `mkstemp'
    /usr/bin/ld: probe_vec.o: in function `ATL_tmpnam':
    /opt/top500/tmp/atlas-build/../ATLAS//CONFIG/include/atlas_sys.h:224: warning: the use of `tmpnam' is dangerous, better use `mkstemp'
    /usr/bin/ld: probe_arch.o: in function `ATL_tmpnam':
    /opt/top500/tmp/atlas-build/../ATLAS//CONFIG/include/atlas_sys.h:224: warning: the use of `tmpnam' is dangerous, better use `mkstemp'
    /usr/bin/ld: /tmp/ccQqmLej.o: in function `ATL_tmpnam':
    /opt/top500/tmp/atlas-build/../ATLAS//CONFIG/include/atlas_sys.h:224: warning: the use of `tmpnam' is dangerous, better use `mkstemp'
    ERROR: enum fam=0, chip=0, model=17, mach=0
    make[3]: *** [Makefile:106: atlas_run] Error 100
    make[2]: *** [Makefile:449: IRunArchInfo_x86] Error 2
    /usr/bin/ld: /tmp/ccwT5Idd.o: in function `ATL_tmpnam':
    /opt/top500/tmp/atlas-build/../ATLAS//CONFIG/include/atlas_sys.h:224: warning: the use of `tmpnam' is dangerous, better use `mkstemp'
    ERROR: enum fam=0, chip=0, model=17, mach=0
    make[3]: *** [Makefile:106: atlas_run] Error 100
    make[2]: *** [Makefile:449: IRunArchInfo_x86] Error 2
    ERROR: enum fam=0, chip=0, model=17, mach=0
    make[3]: *** [Makefile:106: atlas_run] Error 100
    make[2]: *** [Makefile:449: IRunArchInfo_x86] Error 2
    /usr/bin/ld: probe_pmake.o: in function `ATL_tmpnam':
    /opt/top500/tmp/atlas-build/../ATLAS//CONFIG/include/atlas_sys.h:224: warning: the use of `tmpnam' is dangerous, better use `mkstemp'
    ERROR: enum fam=0, chip=0, model=17, mach=0
    make[3]: *** [Makefile:106: atlas_run] Error 100
    make[2]: *** [Makefile:449: IRunArchInfo_x86] Error 2
    It appears you have cpu throttling enabled, which makes timings
    unreliable and an ATLAS install nonsensical.

    OS-controlled CPU throttling is so course grained, that timings become
    essentially random.  What this means for an ATLAS install is that ATLAS
    cannot tell the difference between a good and bad kernel, and so the
    tuning step may result in arbitrarily bad performance.  If you don't care
    about performance, you are usually better off just using the reference BLAS.

    If you fear overheating, setting clock speed to some lower, constant
    value should give you a decent install.

    Hardware-controlled throttling is usually much finer grained, and therefore
    may result in mediocre tuning, but this will depend quite bit on luck.

    If your machine has OS throttling enabled, it is critical that you disable
    it (with something like cpufreq-set).  See INSTALL.txt for details.

    If you you do not care at all about performance, you can rerun configure
    with --cripple-atlas-performance to proceed in the face of throttling.
    Do not do this unless you really don't care about performance.
    If you are able to turn off throttling, rerun configure as normal
    once you have done so.

    Aborting due to throttling
  stderr_lines: <omitted>
  stdout: |-
    gcc -I/opt/top500/tmp/atlas-build/../ATLAS//CONFIG/include  -g -w -c /opt/top500/tmp/atlas-build/../ATLAS//CONFIG/src/atlconf_misc.c
    gcc -I/opt/top500/tmp/atlas-build/../ATLAS//CONFIG/include  -g -w -o xconfig /opt/top500/tmp/atlas-build/../ATLAS//CONFIG/src/config.c atlconf_misc.o
    ./xconfig -d s /opt/top500/tmp/atlas-build/../ATLAS/ -d b /opt/top500/tmp/atlas-build

    OS configured as Linux (1)

    Assembly configured as GAS_x8664 (2)

    Vector ISA Extension configured as  AVXMAC (4,976)

    Architecture configured as  UNKNOWNx86 (43)

    Clock rate configured as 3709Mhz

    Maximum number of threads configured as  384
    Parallel make command configured as '$(MAKE) -j 384'
    CPU Throttling apparently enabled!
    xconfig exited with 1
  stdout_lines: <omitted>
failed: [127.0.0.1] (item=make) => changed=true
  ansible_loop_var: item
  cmd:
  - make
  delta: '0:00:00.003766'
  end: '2022-11-22 06:54:52.880356'
  item: make
  msg: non-zero return code
  rc: 2
  start: '2022-11-22 06:54:52.876590'
  stderr: |-
    make[1]: Make.top: No such file or directory
    make[1]: *** No rule to make target 'Make.top'.  Stop.
    make: *** [Makefile:536: build] Error 2
  stderr_lines: <omitted>
  stdout: |-
    make -f Make.top build
    make[1]: Entering directory '/opt/top500/tmp/atlas-build'
    make[1]: Leaving directory '/opt/top500/tmp/atlas-build'
  stdout_lines: <omitted>

PLAY RECAP ***************************************************************************************************************************************
127.0.0.1                  : ok=10   changed=5    unreachable=0    failed=1    skipped=1    rescued=0    ignored=0

dantemp@genoatest:~/top500-benchmark$
`

Casper042, 2022-11-22:
So slightly confused by this...
1) powerprofilesctl wasn't even installed.
2) This system is running Ubuntu 22 Server, with no GUI, and the article you linked is mainly about incompat with Gnome.
3) Is this the command? `echo "performance" | sudo tee /sys/devices/system/cpu/cpu*/cpufreq/scaling_governor`

Casper042, 2022-11-22:
I changed this line for now, as this machine has beta bios and I'm not sure I can disable any settings easily.

../ATLAS/configure **--cripple-atlas-performance**

geerlingguy, 2022-11-23:
`echo "performance" | sudo tee /sys/devices/system/cpu/cpu*/cpufreq/scaling_governor` <-- Yes... however I think that may have not worked in my simple testing on an Ubuntu 22.04 system. Not sure why.

But see what happens, as long as you're not doing other things on the system while ATLAS compiles, `--cripple-atlas-performance` doesn't do anything too bad.

Casper042, 2022-11-24:
OK, finally got through install but then it bombs on the actual benchmark and I don't understand these errors at all.

```
TASK [Install ATLAS (takes a LONG time).] ********************************************************************************************************
changed: [127.0.0.1] => (item=../ATLAS/configure --cripple-atlas-performance)

changed: [127.0.0.1] => (item=make)

TASK [Create 'COMPILE_ATLAS_COMPLETE' file.] *****************************************************************************************************
changed: [127.0.0.1]

TASK [Download HPL (High Performance Linpack).] **************************************************************************************************
changed: [127.0.0.1]

TASK [Set up HPL makefile.] **********************************************************************************************************************
changed: [127.0.0.1]

TASK [Copy rpi makefile into place.] *************************************************************************************************************
changed: [127.0.0.1]

TASK [Install HPL.] ******************************************************************************************************************************
changed: [127.0.0.1]

TASK [Create COMPILE_HPL_COMPLETE file.] *********************************************************************************************************
changed: [127.0.0.1]

PLAY [Configure SSH connections between nodes.] **************************************************************************************************

PLAY [Run linpack benchmark.] ********************************************************************************************************************

TASK [Gathering Facts] ***************************************************************************************************************************
ok: [127.0.0.1]

TASK [Create a file describing nodes for MPI execution.] *****************************************************************************************
changed: [127.0.0.1]

TASK [Create HPL.dat file.] **********************************************************************************************************************
changed: [127.0.0.1]

TASK [Run the benchmark.] ************************************************************************************************************************
**fatal: [127.0.0.1]: FAILED! => changed=true
  cmd:
  - mpirun
  - -f
  - cluster-hosts
  - ./xhpl
  delta: '0:00:00.054937'
  end: '2022-11-23 22:51:30.048012'
  msg: non-zero return code
  rc: 255
  start: '2022-11-23 22:51:29.993075'
  stderr: |-
    [proxy:0:0@genoatest] HYDU_create_process (utils/launch/launch.c:21): pipe error (Too many open files)
    [proxy:0:0@genoatest] launch_procs (pm/pmiserv/pmip_cb.c:730): create process returned error
    [proxy:0:0@genoatest] HYD_pmcd_pmip_control_cmd_cb (pm/pmiserv/pmip_cb.c:904): launch_procs returned error
    [proxy:0:0@genoatest] HYDT_dmxu_poll_wait_for_event (tools/demux/demux_poll.c:76): callback returned error status
    [proxy:0:0@genoatest] main (pm/pmiserv/pmip.c:169): demux engine error waiting for event
    [mpiexec@genoatest] control_cb (pm/pmiserv/pmiserv_cb.c:206): assert (!closed) failed
    [mpiexec@genoatest] HYDT_dmxu_poll_wait_for_event (tools/demux/demux_poll.c:76): callback returned error status
    [mpiexec@genoatest] HYD_pmci_wait_for_completion (pm/pmiserv/pmiserv_pmci.c:160): error waiting for event
    [mpiexec@genoatest] main (ui/mpich/mpiexec.c:326): process manager error waiting for completion
  stderr_lines: <omitted>
  stdout: ''
  stdout_lines: <omitted>**

NO MORE HOSTS LEFT *******************************************************************************************************************************

PLAY RECAP ***************************************************************************************************************************************
127.0.0.1                  : ok=19   changed=11   unreachable=0    failed=1    skipped=2    rescued=0    ignored=0
```

geerlingguy, 2022-11-27:
It seems like `[proxy:0:0@genoatest] HYDU_create_process (utils/launch/launch.c:21): pipe error (Too many open files)` might be the main issue—too many open files.

Check with https://bobcares.com/blog/too-many-open-files/

closed 2023-06-18 by stale[bot]