geerlingguy/top500-benchmark #9: SSH connection problems running multiple slave nodes on a cluster

lindstedtmj19, 2023-04-11:
Hi Everyone! 

My name is Michael Lindstedt, and I’m a senior student studying Electrical Engineering at Grove City College. I’m working on a project for a parallel computer architecture course with a group of four other people. Our project is to create a supercomputer out of 5 hp laptops and a network switch. We are connecting the laptops as nodes on a cluster on the network and need to run Linpack to measure the performance of the cluster. So far, we removed the windows OS from each computer and replaced it with Linux Ubuntu on each machine. Then, we downloaded the code from this Top500 Benchmark – HPL Linpack Github page. We used one laptop as the master node distributing the workload, and the other four as slave nodes receiving the workload. When we run the program per the instructions, we can get the benchmark to run successfully with one master node and one slave node. However, when we run the benchmark with multiple slave nodes, one or more nodes always fail while attempting to establish ssh connections after we are prompted to enter the passcode for the host node several times in command line. We’ve tried several options including modifying some of the relevant files and researching how changing those files might help, but we haven’t had any luck. We need to finish the project by Thursday April 20. Has anyone ever encountered this error before when attempting to benchmark a cluster with this code? Do you have any suggestions on how we might resolve this error? Any help would be greatly appreciated!

Thanks in advance,

Mike Lindstedt

geerlingguy, 2023-04-11:
One thing that immediately comes to mind is the DNS resolution to all hostnames.

Make sure they are all unique, and then verify the `/etc/hosts` file on each of the machines. See this task, which should put the hostnames in place: https://github.com/geerlingguy/top500-benchmark/blob/master/main.yml#L182-L189

Without the hostnames all correct and unique, you could run into MPI not being able to distinguish one node from another.

lindstedtmj19, 2023-04-12:
Thank you. I'll give try that later this week and let you know if that fixes it.

benprog19, 2023-04-17:
Hi, from the same team,
We have configured out hosts.ini file and our config.yml file to what we believe to be the correct format for our system. We have 4 laptops with 4 cores each. So this is our resulting config.yml file:
`---
# Working directory where HPL and associated applications will be compiled.
hpl_root: /home/ben/Documents/top500-benchmark

# Home directory of the user for whom SSH keys will be configured
ssh_user: ben
ssh_user_home: /home/ben

# Specify manually if needed for mixed-RAM-capacity clusters.
ram_in_gb: "{{ ( ansible_memtotal_mb / 1024 * 0.75 ) | int | abs }}"

# Count the nodes for accurate HPL.dat calculations.
nodecount: "{{ ansible_play_hosts | length | int }}"

# HPL.dat configuration options.
# See: https://www.advancedclustering.com/act_kb/tune-hpl-dat-file/
# See also: https://hpl-calculator.sourceforge.net/HPL-HowTo.pdf
hpl_dat_opts:
  # sqrt((Memory in GB * 1024 * 1024 * 1024 * Node count) / 8) * 0.9
  Ns: "{{ (((((ram_in_gb | int) * 1024 * 1024 * 1024 * (nodecount | int)) / 8) | root) * 0.90) | int }}"
  NBs: 256
  # (P * Q) should be roughly equivalent to total core count, with Qs higher.
  # If running on a single system, Ps should be 1 and Qs should be core count.
  Ps: 4
  Qs: 4`

Our hosts.ini file has this format:
`# For single node benchmarking (default), use this:
#[cluster]
#127.0.0.1 ansible_connection=local

# For cluster benchmarking, delete everything above this line and uncomment:
[cluster]
ben@node1	ansible_connection=local
ben@node2	ansible_connection=local
michael@node3	ansible_connection=local
ben@node4	ansible_connection=local

[cluster:vars]
ansible_user=ben
#ansible_user=michael
#ansible_user=ben`

And our resulting /etc/hosts file after running the benchmark is:
`127.0.0.1	localhost
127.0.1.1	headnode

10.15.105.6	node1
10.15.105.11	node2
10.15.107.212	node3
10.15.106.141	node4

# The following lines are desirable for IPv6 capable hosts
::1     ip6-localhost ip6-loopback
fe00::0 ip6-localnet
ff00::0 ip6-mcastprefix
ff02::1 ip6-allnodes
ff02::2 ip6-allrouters

# BEGIN Ansible MPI host ben@node1
10.15.105.147 ben@node1 ben@node1
# END Ansible MPI host ben@node1
# BEGIN Ansible MPI host ben@node2
10.15.105.147 ben@node2 ben@node2
# END Ansible MPI host ben@node2
# BEGIN Ansible MPI host michael@node3
10.15.105.147 michael@node3 michael@node3
# END Ansible MPI host michael@node3
[... 68 lines cut ...]
    - The relative machine precision (eps) is taken to be               1.110223e-16
    - Computational tests pass if scaled residuals are less than                16.0
  
  
    ===================================================================================
    =   BAD TERMINATION OF ONE OF YOUR APPLICATION PROCESSES
    =   PID 26292 RUNNING AT 10.15.105.147
    =   EXIT CODE: 9
    =   CLEANING UP REMAINING PROCESSES
    =   YOU CAN IGNORE THE BELOW CLEANUP MESSAGES
    ===================================================================================
    YOUR APPLICATION TERMINATED WITH THE EXIT STRING: Killed (signal 9)
    This typically refers to a problem with your application.
    Please see the FAQ page for debugging suggestions
  stdout_lines: <omitted>

NO MORE HOSTS LEFT ****************************************************************************************************************************************************************************************

PLAY RECAP ************************************************************************************************************************************************************************************************
ben@node1                  : ok=26   changed=6    unreachable=0    failed=1    skipped=3    rescued=0    ignored=0   
ben@node2                  : ok=25   changed=5    unreachable=0    failed=0    skipped=3    rescued=0    ignored=0   
ben@node4                  : ok=25   changed=6    unreachable=0    failed=0    skipped=3    rescued=0    ignored=0   
michael@node3              : ok=25   changed=6    unreachable=0    failed=0    skipped=3    rescued=0    ignored=0   `

Any suggestions to fix this? Thanks!

geerlingguy, 2023-04-17:
@benprog19 - Those hostnames seem invalid to me. I'm assuming `ben@node1` means that the username is `ben` and the hostname is `node1`?

If that's the case, you should set the hostname as `node`, and then on that line put `ansible_user=ben`, e.g.:

```
[cluster]
node1	ansible_connection=local ansible_user=ben
node2	ansible_connection=local ansible_user=ben
node3	ansible_connection=local ansible_user=michael
node4	ansible_connection=local ansible_user=ben
```

benprog19, 2023-04-18:
@geerlingguy, we changed the /etc/hosts files on each node and the hosts.ini file to match correctly. We are running the "sudo ansible_playbook main.yml" command with 2 nodes and it seems to be running everything on the headnode. Our system monitor starts spiking and the ram climbs to max ram usage and then drops with an out of memory error. It should be running on the two other nodes correct? What would cause it to run on the headnode instead of the individual nodes?

benprog19, 2023-04-18:
**Node 1 System Monitor during Run**
[image]

**Node 2 System Monitor during Run**
[image]

**HeadNode System Monitor during Run**
[image]

As you can see, the headnode is running the benchmark and slowly climbing to max memory. When it reaches max memory it crashes the benchmark with the error code 9 memory error.

geerlingguy, 2023-04-18:
Do all of your systems (including the head node) have the same amount of RAM? The way the HPL.dat file is structured, it's meant to run on a cluster where all nodes are identical. You might need to make some changes if some have less RAM.

benprog19, 2023-04-18:
They all have the same specs.

geerlingguy, 2023-04-18:
In that case, maybe adjust the `ram_in_gb` value down a bit — instead of * 0.75, change that * 0.60 or even less to make sure it doesn't fill up the RAM.

benprog19, 2023-04-18:
> In that case, maybe adjust the `ram_in_gb` value down a bit — instead of * 0.75, change that * 0.60 or even less to make sure it doesn't fill up the RAM. (see: https://github.com/geerlingguy/top500-benchmark/blob/master/example.config.yml#L10)

Will do, will that fix the problem where the other nodes aren't being started running in parallel?

geerlingguy, 2023-04-18:
@benprog19 - No, that problem would generally be caused by the hosts file being incorrect, did you make the change in https://github.com/geerlingguy/top500-benchmark/issues/9#issuecomment-1512117249 and what does your hosts file look like now?

benprog19, 2023-04-18:
Yes we did make the changes:

lindstedtmj19, 2023-04-19:
@geerlingguy based on our set up in the image Ben sent in the last comment in this thread, is our hosts file configured correctly?

geerlingguy, 2023-04-19:
Yes — though the way it is set up in that picture, only the two first nodes will be connected. Make sure to uncomment node3 and node4 if you want those in the mix.

benprog19, 2023-04-19:
We were just testing with two nodes to see if that worked, but it still just ran on the headnode and didn't run anything on the other two machines while looking at the system monitor.

geerlingguy, 2023-04-19:
Hmm... right now I'm a bit out of ideas, but every time I ran into similar issues it turned out to be a DNS issue. If mDNS is not working on your network, or the computers are plugged into different network segments (or on different WiFi access points), then it could be the case that the simple hostnames aren't working through your network with HPL. See: https://github.com/geerlingguy/top500-benchmark/issues/1#issuecomment-1317337367

closed 2023-10-26 by github-actions[bot]