geerlingguy/top500-benchmark #14: Benchmark Orange Pi 5

geerlingguy, 2023-09-05:
The Orange Pi 5 has a Rockchip RK3588S with 8 cores (A76 quad core + A55 quad core), and runs up to 2.4 GHz.

geerlingguy, 2023-09-07:
Using `blis`:

```
================================================================================
HPLinpack 2.3  --  High-Performance Linpack benchmark  --   December 2, 2018
Written by A. Petitet and R. Clint Whaley,  Innovative Computing Laboratory, UTK
Modified by Piotr Luszczek, Innovative Computing Laboratory, UTK
Modified by Julien Langou, University of Colorado Denver
================================================================================

An explanation of the input/output parameters follows:
T/V    : Wall time / encoded variant.
N      : The order of the coefficient matrix A.
NB     : The partitioning blocking factor.
P      : The number of process rows.
Q      : The number of process columns.
Time   : Time in seconds to solve the linear system.
Gflops : Rate of execution for solving the linear system.

The following parameter values will be used:

N      :   14745
NB     :     256
PMAP   : Row-major process mapping
P      :       1
Q      :       4
PFACT  :   Right
NBMIN  :       4
NDIV   :       2
RFACT  :   Crout
BCAST  :  1ringM
DEPTH  :       1
SWAP   : Mix (threshold = 64)
L1     : transposed form
U      : transposed form
EQUIL  : yes
ALIGN  : 8 double precision words

--------------------------------------------------------------------------------

- The matrix A is randomly generated for each test.
- The following scaled residual check will be computed:
      ||Ax-b||_oo / ( eps * ( || x ||_oo * || A ||_oo + || b ||_oo ) * N )
- The relative machine precision (eps) is taken to be               1.110223e-16
- Computational tests pass if scaled residuals are less than                16.0

================================================================================
T/V                N    NB     P     Q               Time                 Gflops
--------------------------------------------------------------------------------
WR11C2R4       14745   256     1     4              40.08             5.3333e+01
HPL_pdgesv() start time Thu Sep  7 14:28:27 2023

HPL_pdgesv() end time   Thu Sep  7 14:29:07 2023

--------------------------------------------------------------------------------
||Ax-b||_oo/(eps*(||A||_oo*||x||_oo+||b||_oo)*N)=   4.07780599e-03 ...... PASSED
================================================================================

Finished      1 tests with the following results:
              1 tests completed and passed residual checks,
              0 tests completed and failed residual checks,
              0 tests skipped because of illegal input values.
--------------------------------------------------------------------------------

End of Tests.
================================================================================
```

Results:
  - `1` Ps, `4` Qs (above): Score of 53.33 Gflops using 11.5W average, 11.7W peak
  - `1` Ps, `8` Qs: Score of 37.143 Gflops using 12.4W average, 13W peak

closed 2023-09-07 by geerlingguy

dbohdan, 2025-01-09:
Review: https://github.com/geerlingguy/sbc-reviews/issues/5.