geerlingguy/top500-benchmark #22: Benchmark Turing Machines RK1 (RK3588 octo-core)

geerlingguy, 2023-11-03:
Turing Machines RK1 has an 8-core RK3588 onboard, but is a new System on Module design. The first one I've seen come to market (should be shipping this month). See my full workup on it here: https://github.com/geerlingguy/sbc-reviews/issues/25

geerlingguy, 2023-11-03:
```
================================================================================
HPLinpack 2.3  --  High-Performance Linpack benchmark  --   December 2, 2018
Written by A. Petitet and R. Clint Whaley,  Innovative Computing Laboratory, UTK
Modified by Piotr Luszczek, Innovative Computing Laboratory, UTK
Modified by Julien Langou, University of Colorado Denver
================================================================================

An explanation of the input/output parameters follows:
T/V    : Wall time / encoded variant.
N      : The order of the coefficient matrix A.
NB     : The partitioning blocking factor.
P      : The number of process rows.
Q      : The number of process columns.
Time   : Time in seconds to solve the linear system.
Gflops : Rate of execution for solving the linear system.

The following parameter values will be used:

N      :   34581
NB     :     256
PMAP   : Row-major process mapping
P      :       1
Q      :       8
PFACT  :   Right
NBMIN  :       4
NDIV   :       2
RFACT  :   Crout
BCAST  :  1ringM
DEPTH  :       1
SWAP   : Mix (threshold = 64)
L1     : transposed form
U      : transposed form
EQUIL  : yes
ALIGN  : 8 double precision words

--------------------------------------------------------------------------------

- The matrix A is randomly generated for each test.
- The following scaled residual check will be computed:
      ||Ax-b||_oo / ( eps * ( || x ||_oo * || A ||_oo + || b ||_oo ) * N )
- The relative machine precision (eps) is taken to be               1.110223e-16
- Computational tests pass if scaled residuals are less than                16.0

================================================================================
T/V                N    NB     P     Q               Time                 Gflops
--------------------------------------------------------------------------------
WR11C2R4       34581   256     1     8             539.46             5.1108e+01
HPL_pdgesv() start time Fri Nov  3 16:50:26 2023

HPL_pdgesv() end time   Fri Nov  3 16:59:25 2023

--------------------------------------------------------------------------------
||Ax-b||_oo/(eps*(||A||_oo*||x||_oo+||b||_oo)*N)=   2.33869652e-03 ...... PASSED
================================================================================

Finished      1 tests with the following results:
              1 tests completed and passed residual checks,
              0 tests completed and failed residual checks,
              0 tests skipped because of illegal input values.
--------------------------------------------------------------------------------

End of Tests.
================================================================================
```

I can't yet get a power reading, as I am running the thing on a Turing Pi 2. I wasn't comfortable with how it fit on my Jenson Nano Dev Board (the retention clips hit a couple circuits on it).

geerlingguy, 2023-11-03:
With P=1 and Q=8 (matching up some testing I did on the Rock 5 B):

```
TODO
```

geerlingguy, 2024-02-23:
The above results were taken with a 16GB RAM version, I'm going to re-test with a 32GB version now.

geerlingguy, 2024-02-23:
59.581 Gflops (unknown wattage, see https://github.com/geerlingguy/sbc-reviews/issues/38 for more — will try to get some sort of power reading soon).

closed 2024-06-27 by geerlingguy

UnknownNet, 2024-12-17:
Was going through your RK1 single-node results (specifically for the 32G model) and had found some discrepencies. Turns out that openmpi by default (no proc mapping options) will default to cores 0-3 (Cortex-A55) while mpich defaults to cores 4-7 (Cortex-A76). Setting "--bind-to l3cache" with openmpi's mpirun it will push to using cores 4-7 and results are within noise of your posted results. This is also why setting PxQ = 4 results in faster timings than PxQ = 8 (for all available cores) as HPL will be throttled by the slower cores.