geerlingguy/top500-benchmark #28: Ampere Altra Max M128-30 slower than lower models?

alexVinarskis, 2024-03-16:
Was wondering, since it seems both M128-28 and M96-28 outperform M128-30, while pulling significantly less power. Could it have to do with different test version (M128-30 was tested much earlier than others), or is it an actual known chip limitation?

geerlingguy, 2024-03-16:
@alexVinarskis - That earlier test was using a much older version of this repo, which didn't include any optimizations for modern Arm chips. I was considering taking that out of the listing, mostly due to that older setup, but I have some good information in that original issue though...

alexVinarskis, 2024-03-16:
@geerlingguy thanks for the fast reply, that perfectly answers it.

Considering setting up a server with M128-30, got worried whether a lower spec chip would somehow be faster :) If I go through with it, might be able to contribute with an up-to date test results then.

closed 2024-03-16 by alexVinarskis

geerlingguy, 2024-03-18:
@alexVinarskis - Yes! If you do end up with access to that chip, I'd love to get a better test result.