geerlingguy/top500-benchmark #36: Benchmark Raspberry Pi 5 Linux kernel NUMA patch

geerlingguy, 2024-07-06:
I would like to see if the NUMA patch here: https://lore.kernel.org/lkml/20240625125803.38038-1-tursulin@igalia.com/ — has any bearing on HPL performance and/or efficiency scores. Especially if it's reproducible and significant.

The stated numbers for Geekbench 6 are 5-ish and 20-ish percent improvements for single/multicore. I would like to see if there's any impact for HPL (which is inherently multicore, and very RAM-speed-dependent). Also measure the power usage to see if this affects power draw positively, negatively, or not at all.

geerlingguy, 2024-07-06:
## Baseline

  - Idle: 3.5W
  - Benchmark: TODOW

Result:

```
TODO: ansible-playbook main.yml --tags "setup,benchmark"
```

geerlingguy, 2024-07-07:
## After applying NUMA patch

  - Idle: TODOW
  - Benchmark: TODOW
  - Result: TODO Gflops
  - Efficiency: TODO Gflops/W
  - Geekbench: TODO single / TODO multi ([link](TODO))

Result:

```
TODO
```

geerlingguy, 2024-07-07:
I also ran Geekbench 6 just after boot (1 min uptime) with the NUMA patch in place. Here's the result: https://browser.geekbench.com/v6/cpu/6820837 (801 / 1636).

geerlingguy, 2024-07-07:
And another Geekbench 6 run about 1 hour after boot, after cooldown period of 10 minutes after all the previous tests: https://browser.geekbench.com/v6/cpu/6821505 (799 / 1637). So no noticeable difference at least on this Pi 5 8GB running this Linux kernel between runs immediately following boot and runs much later.

Going to move some other performance testing over to https://github.com/geerlingguy/sbc-reviews/issues/21

will127534, 2024-07-08:
Hi @geerlingguy, I'm running through your steps and I think we also need to add `numa=fake=4` to the cmdline.txt.

geerlingguy, 2024-07-08:
@will127534 - heh... as I was writing up a bit of a post on this... I realized that exact step was missing. I'm going to re-test now. Adding `numa=fake=4` to `/boot/firmware/config.txt` and rebooting, I now see:

```
pi@pi5:~ $ dmesg
...
[    0.000000] NUMA: No NUMA configuration found
[    0.000000] Faking a node at [mem 0x0000000000000000-0x000000007fffffff]
[    0.000000] Faking a node at [mem 0x0000000080000000-0x00000000ffffffff]
[    0.000000] Faking a node at [mem 0x0000000100000000-0x000000017fffffff]
[    0.000000] Faking a node at [mem 0x0000000180000000-0x00000001ffffffff]
...
[    0.000000] Kernel command line: reboot=w coherent_pool=1M 8250.nr_uarts=1 pci=pcie_bus_safe  smsc95xx.macaddr=D8:3A:DD:84:FB:3A vc_mem.mem_base=0x3fc00000 vc_mem.mem_size=0x40000000  console=ttyAMA10,115200 console=tty1 root=PARTUUID=9f1af6e7-02 rootfstype=ext4 fsck.repair=yes numa=fake=4 rootwait
```

closed 2024-07-11 by geerlingguy