geerlingguy/top500-benchmark #43: Benchmark AmpereOne A192-32X

geerlingguy, 2024-10-16:
I have a system Ampere / Supermicro sent over for testing, it's the 192-core CPU, with 512 GB of DDR5 RAM at 5200 MT/s.

geerlingguy, 2024-10-16:
CPU was self-reporting 297W power draw, I don't have my wall power measurement set up yet, but will soon. The power strip showed 8A at 120V (for the full rack), so somewhere between 500-600W for the full system probably.

```
================================================================================
HPLinpack 2.3  --  High-Performance Linpack benchmark  --   December 2, 2018
Written by A. Petitet and R. Clint Whaley,  Innovative Computing Laboratory, UTK
Modified by Piotr Luszczek, Innovative Computing Laboratory, UTK
Modified by Julien Langou, University of Colorado Denver
================================================================================

An explanation of the input/output parameters follows:
T/V    : Wall time / encoded variant.
N      : The order of the coefficient matrix A.
NB     : The partitioning blocking factor.
P      : The number of process rows.
Q      : The number of process columns.
Time   : Time in seconds to solve the linear system.
Gflops : Rate of execution for solving the linear system.

The following parameter values will be used:

N      :  203788
NB     :     256
PMAP   : Row-major process mapping
P      :       1
Q      :     192
PFACT  :   Right
NBMIN  :       4
NDIV   :       2
RFACT  :   Crout
BCAST  :  1ringM
DEPTH  :       1
SWAP   : Mix (threshold = 64)
L1     : transposed form
U      : transposed form
EQUIL  : yes
ALIGN  : 8 double precision words

--------------------------------------------------------------------------------

- The matrix A is randomly generated for each test.
- The following scaled residual check will be computed:
      ||Ax-b||_oo / ( eps * ( || x ||_oo * || A ||_oo + || b ||_oo ) * N )
- The relative machine precision (eps) is taken to be               1.110223e-16
- Computational tests pass if scaled residuals are less than                16.0

================================================================================
T/V                N    NB     P     Q               Time                 Gflops
--------------------------------------------------------------------------------
WR11C2R4      203788   256     1   192            2628.74             2.1464e+03
HPL_pdgesv() start time Wed Oct 16 17:08:19 2024

HPL_pdgesv() end time   Wed Oct 16 17:52:08 2024

--------------------------------------------------------------------------------
||Ax-b||_oo/(eps*(||A||_oo*||x||_oo+||b||_oo)*N)=   2.21811736e-03 ...... PASSED
================================================================================

Finished      1 tests with the following results:
              1 tests completed and passed residual checks,
              0 tests completed and failed residual checks,
              0 tests skipped because of illegal input values.
--------------------------------------------------------------------------------

End of Tests.
================================================================================
```

geerlingguy, 2024-10-16:
That score would put this system at rank #460 on the 2006/06 Top500 List (see https://hpl-calculator.sourceforge.net/hpl-calculations.php and https://www.top500.org/lists/top500/list/2006/06/?page=5).

It about matches the [Saguaro cluster at Arizona State University](https://www.top500.org/system/174269/), which consumed around 80 kilowatts of power in 2006.

geerlingguy, 2024-10-25:
Re-ran the test again today with my ThirdReality Smart Outlet for power monitoring. It is within 10W of Supermicro's BMC power reporting. See image and full results below, but here's the summary:

TODO Gflops at TODO W for TODO Gflops/W

[image]

```
TODO
```

Ampere is going to provide instructions for a more optimized HPL run since AmpereOne is a newer architecture and could use a few optimizations to sneak in a bit more efficiency.

rbapat-ampere, 2024-10-25:
Hi Jeff, 
Can you rerun the workload using the HPL.dat input file attached?
Our testing shows us getting 3.0 TFLOP/s 
[HPL.zip](https://github.com/user-attachments/files/17527362/HPL.zip)

Note: We are working on tweaking the performance even further.

geerlingguy, 2024-10-25:
@rbapat-ampere - Sure! Makes sense if the NUMA layout means the memory access is better allocated like that. It seems highly dependent on the core layout, on most single-socket systems just having the number of cores for Qs helps, but when we hit 192 cores, things behave funny :)

I'm going to use the following values in my `config.yml`, to override the defaults (and I was using 1/192 Ps/Qs):

```yaml
hpl_dat_opts:
  # sqrt((Memory in GB * 1024 * 1024 * 1024 * Node count) / 8) * 0.9
  #Ns: "{{ (((((ram_in_gb | int) * 1024 * 1024 * 1024 * (nodecount | int)) / 8) | root) * 0.90) | int }}"
  Ns: 203788
  NBs: 256
  # (P * Q) should be roughly equivalent to total core count, with Qs higher.
  # If running on a single system, Ps should be 1 and Qs should be core count.
  Ps: 12
  Qs: 16 
```

geerlingguy, 2024-10-25:
TODO Gflops at 685 W, for TODO Gflops/W

Average is closer to 685W on this run, with the above settings:

[image]

Final result:

```
TODO
```

rbapat-ampere, 2024-10-27:
@geerlingguy 
Nothing of that sort. Used OpenBLAS. 

```shell
================================================================================
T/V                N    NB     P     Q               Time                 Gflops
--------------------------------------------------------------------------------
WR11C2R4      239872   256    12    16            3095.07             2.9729e+03
HPL_pdgesv() start time Fri Oct 25 10:23:52 2024

HPL_pdgesv() end time   Fri Oct 25 11:15:27 2024

--------------------------------------------------------------------------------
||Ax-b||_oo/(eps*(||A||_oo*||x||_oo+||b||_oo)*N)=   1.50966052e-03 ...... PASSED
================================================================================
```

geerlingguy, 2024-10-28:
@rbapat-ampere / @joespeed - I noticed your `N` was set to `239872` in the run above (the HPL.zip you posted had `N = 203788`).

I will do another run with that updated value.

The entire build procedure is done automatically in an isolated build directory using this very project's Ansible playbook: https://github.com/geerlingguy/top500-benchmark

The build instructions are:

  1. Ensure you have Ansible installed locally (e.g. `pip install ansible`)
  2. Clone this repository anywhere on the local volume
  3. `cp example.hosts.ini hosts.ini && cp example.config.yml config.yml`
  4. Modify the `config.yml` file and set the exact values you require (my entire `config.yml` is pasted below)
  5. Run `ansible-playbook main.yml --tags "setup,benchmark"`

The Ansible playbook makes it easy for me to swap out libraries, re-run the tests dozens of times without having to touch anything, and verify that my test runs are _exactly_ the same across systems and time (as I also control the versions of libraries I install).

In my previous runs I've used flame/blis, but I will switch to `openblas`. The mpich version I'm running right now is `4.2.2`, but I see `4.2.3` was just released, so I will switch to that as well. I'm starting a new run momentarily.

rbapat-ampere, 2024-10-28:
@geerlingguy 
I ran HPL using your top500 ansible playbook. A few things I noticed during runtime.

1. While checking htop, I noticed that the average all core utilization did not reach 100%
[image]

During my runs, its 100% from get go all the way to the end.

geerlingguy, 2024-10-28:
@rbapat-ampere - To match your setup, you would need to change the `config.yml` values to match the ones I pasted in my comment (I forgot to paste initially, but they are there now).

I just re-ran the benchmark and it errored out. Checking dmesg, I found a bunch of errors like:

```
[  511.304636] EDAC MC0: 1 CE single-symbol chipkill ECC on P0_Node0_Channel5_Dimm0 DIMMF1 (node:0 card:5 module:16 rank:0 bank_group:3 bank_address:3 device:7 row:32680 column:1312 DIMM location: P0_Node0_Channel5_Dimm0 DIMMF1 page:0x3fd437 offset:0x4180 grain:1 syndrome:0x0 - APEI location: node:0 card:5 module:16 rank:0 bank_group:3 bank_address:3 device:7 row:32680 column:1312 DIMM location: P0_Node0_Channel5_Dimm0 DIMMF1 status(0x0000000000000400): Storage error in DRAM memory)
```

And at the end, as I got an error from HPL, I saw this in `dmesg`:

```
[  536.679810] EDAC MC0: 1 CE single-symbol chipkill ECC on P0_Node0_Channel5_Dimm0 DIMMF1 (node:0 card:5 module:16 rank:0 bank_group:3 bank_address:3 device:7 row:1492 column:1248 DIMM location: P0_Node0_Channel5_Dimm0 DIMMF1 page:0x2ea37 offset:0x3080 grain:1 syndrome:0x0 - APEI location: node:0 card:5 module:16 rank:0 bank_group:3 bank_address:3 device:7 row:1492 column:1248 DIMM location: P0_Node0_Channel5_Dimm0 DIMMF1 status(0x0000000000000400): Storage error in DRAM memory)
[  539.061145] {139}[Hardware Error]: Hardware error from APEI Generic Hardware Error Source: 2
[  539.061148] {139}[Hardware Error]: It has been corrected by h/w and requires no further action
[  539.061149] {139}[Hardware Error]: event severity: corrected
[  539.061150] {139}[Hardware Error]:  Error 0, type: corrected
[  539.061151] {139}[Hardware Error]:   section_type: memory error
[  539.061152] {139}[Hardware Error]:    error_status: Storage error in DRAM memory (0x0000000000000400)
[  539.061154] {139}[Hardware Error]:   physical_address: 0x00000001b0b63bc0
[  539.061156] {139}[Hardware Error]:   node:0 card:5 module:16 rank:0 bank_group:3 bank_address:3 device:7 row:865 column:240 
[  539.061157] {139}[Hardware Error]:   error_type: 4, single-symbol chipkill ECC
[  539.061158] {139}[Hardware Error]:   DIMM location: P0_Node0_Channel5_Dimm0 DIMMF1 
[  539.061159] {139}[Hardware Error]:  Error 1, type: corrected
[  539.061160] {139}[Hardware Error]:   section type: unknown, 2826cc9f-448c-4c2b-86b6-a95394b7ef33
[  539.061161] {139}[Hardware Error]:   section length: 0x30
[  539.061163] {139}[Hardware Error]:   00000000: 00003001 00000015 1b0b63c0 00000000  .0.......c......
[  539.061165] {139}[Hardware Error]:   00000010: 00000000 00000000 00000000 00410002  ..............A.
[  539.061167] {139}[Hardware Error]:   00000020: 001d0000 00000002 0d600080 006e0000  ..........`...n.
[  539.061170] EDAC MC0: 1 CE single-symbol chipkill ECC on P0_Node0_Channel5_Dimm0 DIMMF1 (node:0 card:5 module:16 rank:0 bank_group:3 bank_address:3 device:7 row:865 column:240 DIMM location: P0_Node0_Channel5_Dimm0 DIMMF1 page:0x1b0b6 offset:0x3bc0 grain:1 syndrome:0x0 - APEI location: node:0 card:5 module:16 rank:0 bank_group:3 bank_address:3 device:7 row:865 column:240 DIMM location: P0_Node0_Channel5_Dimm0 DIMMF1 status(0x0000000000000400): Storage error in DRAM memory)
```

Indeed, as I'm running the test, I'm seeing thousands of these messages every second...

```
[  669.816722] EDAC MC0: 1 CE single-symbol chipkill ECC on P0_Node0_Channel5_Dimm0 DIMMF1 (node:0 card:5 module:16 rank:0 bank_group:3 bank_address:3 device:7 row:43119 column:1776 DIMM location: P0_Node0_Channel5_Dimm0 DIMMF1 page:0x5437b7 offset:0xb8c0 grain:1 syndrome:0x0 - APEI location: node:0 card:5 module:16 rank:0 bank_group:3 bank_address:3 device:7 row:43119 column:1776 DIMM location: P0_Node0_Channel5_Dimm0 DIMMF1 status(0x0000000000000400): Storage error in DRAM memory)
```

It seems like `DIMM location: P0_Node0_Channel5_Dimm0 DIMMF1` might be a bad stick of RAM (and would account for a slower result...).

geerlingguy, 2024-10-28:
Indeed, looking in the BMC, I see a bunch of health warnings:

[image]

geerlingguy, 2024-10-28:
Aww, this time it ran for 30 minutes before bailing with OOM...

```
    ===================================================================================
    =   BAD TERMINATION OF ONE OF YOUR APPLICATION PROCESSES
    =   PID 41523 RUNNING AT 10.0.2.21
    =   EXIT CODE: 9
    =   CLEANING UP REMAINING PROCESSES
    =   YOU CAN IGNORE THE BELOW CLEANUP MESSAGES
    ===================================================================================
```

Going to give it one more go and then maybe back off `N`s a bit more. All 8 sticks of RAM accounted for, 512 GB total system RAM:

```
ubuntu@ubuntu:~$ free -h
               total        used        free      shared  buff/cache   available
Mem:           510Gi       397Gi       131Gi       6.5Mi       450Mi       113Gi
Swap:          8.0Gi       166Mi       7.8Gi

ubuntu@ubuntu:~$ sudo dmidecode -t memory | grep Serial
	Serial Number: 80AD012324466511C5
	Serial Number: 80AD012324466511CE
	Serial Number: 80AD01241097E76174
	Serial Number: 80AD01232446651180
	Serial Number: 80AD01232446651279
	Serial Number: 80AD01232446651347
	Serial Number: 80AD012327879F620A
	Serial Number: 80AD012327879F624D
```

...maybe OpenBLAS handles memory allocation differently? We'll see if a 2nd run works or if it errors out at the same point. Note that power usage is _dramatically_ reduced, more like 430W. I can't get a result yet, though, so we'll see what that means for overall efficiency!

geerlingguy, 2024-10-28:
After 30 minutes, it looks like free space does run out, and swap is touched... so probably a good idea to limit the RAM a little further.

[image]

TODO...

geerlingguy, 2024-10-28:
Even at `0.25`x the system RAM, it seems like the RAM fills up around 30 minutes into the test. Might switch back to `blis` and see if the behavior improves again...

@rbapat-ampere - When you're doing your runs, how long do they take? And have you monitored RAM consumption over time to see if/when it fills up with 512 GB of system memory?

geerlingguy, 2024-10-28:
End result with `blis` again:

```
================================================================================
HPLinpack 2.3  --  High-Performance Linpack benchmark  --   December 2, 2018
Written by A. Petitet and R. Clint Whaley,  Innovative Computing Laboratory, UTK
Modified by Piotr Luszczek, Innovative Computing Laboratory, UTK
Modified by Julien Langou, University of Colorado Denver
================================================================================

An explanation of the input/output parameters follows:
T/V    : Wall time / encoded variant.
N      : The order of the coefficient matrix A.
NB     : The partitioning blocking factor.
P      : The number of process rows.
Q      : The number of process columns.
Time   : Time in seconds to solve the linear system.
Gflops : Rate of execution for solving the linear system.

The following parameter values will be used:

N      :  203788
NB     :     256
PMAP   : Row-major process mapping
P      :      12
Q      :      16
PFACT  :   Right
NBMIN  :       4
NDIV   :       2
RFACT  :   Crout
BCAST  :  1ringM
DEPTH  :       1
SWAP   : Mix (threshold = 64)
L1     : transposed form
U      : transposed form
EQUIL  : yes
ALIGN  : 8 double precision words

--------------------------------------------------------------------------------

- The matrix A is randomly generated for each test.
- The following scaled residual check will be computed:
      ||Ax-b||_oo / ( eps * ( || x ||_oo * || A ||_oo + || b ||_oo ) * N )
- The relative machine precision (eps) is taken to be               1.110223e-16
- Computational tests pass if scaled residuals are less than                16.0

================================================================================
T/V                N    NB     P     Q               Time                 Gflops
--------------------------------------------------------------------------------
WR11C2R4      203788   256    12    16            2624.94             2.1495e+03
HPL_pdgesv() start time Mon Oct 28 20:35:19 2024

HPL_pdgesv() end time   Mon Oct 28 21:19:04 2024

--------------------------------------------------------------------------------
||Ax-b||_oo/(eps*(||A||_oo*||x||_oo+||b||_oo)*N)=   1.71446644e-03 ...... PASSED
================================================================================

Finished      1 tests with the following results:
              1 tests completed and passed residual checks,
              0 tests completed and failed residual checks,
              0 tests skipped because of illegal input values.
--------------------------------------------------------------------------------

End of Tests.
================================================================================
```

geerlingguy, 2024-10-28:
I tested again, just with 0.05 of the RAM (so `N = 52133`), and this time it averaged around 450W of power.

TODO, waiting for test to complete...

geerlingguy, 2024-10-28:
On `v0.3.28`, I'm able to get a run to complete setting `N` laughably low. Setting it to `166501` it seems like it may be working better, using up 99% of RAM but not thrashing... We'll see.

geerlingguy, 2024-10-29:
That test failed with more:

```
[26986.597698] oom-kill:constraint=CONSTRAINT_NONE,nodemask=(null),cpuset=user.slice,mems_allowed=0,global_oom,task_memcg=/user.slice/user-1000.slice/user@1000.service/init.scope,task=systemd,pid=529208,uid=1000
[26986.597752] Out of memory: Killed process 529208 (systemd) total-vm:22080kB, anon-rss:0kB, file-rss:0kB, shmem-rss:0kB, UID:1000 pgtables:384kB oom_score_adj:100
[26986.640893] xhpl invoked oom-killer: gfp_mask=0x100cca(GFP_HIGHUSER_MOVABLE), order=0, oom_score_adj=0
```

So I'm switching gears and will run the scripts sent over from @rbapat-ampere to compile the versions of OpenMPI 5.0.5, BLIS and HPL he has running. I believe a repo with that configuration will be forthcoming...

geerlingguy, 2024-10-29:
Finally got a good run in, with the Ampere-supplied script.

2,996 Gflops at 725W, for 4.13 Gflops/W

```
================================================================================
HPLinpack 2.3  --  High-Performance Linpack benchmark  --   December 2, 2018
Written by A. Petitet and R. Clint Whaley,  Innovative Computing Laboratory, UTK
Modified by Piotr Luszczek, Innovative Computing Laboratory, UTK
Modified by Julien Langou, University of Colorado Denver
================================================================================

An explanation of the input/output parameters follows:
T/V    : Wall time / encoded variant.
N      : The order of the coefficient matrix A.
NB     : The partitioning blocking factor.
P      : The number of process rows.
Q      : The number of process columns.
Time   : Time in seconds to solve the linear system.
Gflops : Rate of execution for solving the linear system.

The following parameter values will be used:

N      :  239872 
NB     :     256 
PMAP   : Row-major process mapping
P      :      12 
Q      :      16 
PFACT  :   Right 
NBMIN  :       4 
NDIV   :       2 
RFACT  :   Crout 
BCAST  :  1ringM 
DEPTH  :       1 
SWAP   : Mix (threshold = 64)
L1     : transposed form
U      : transposed form
EQUIL  : yes
ALIGN  : 8 double precision words

--------------------------------------------------------------------------------

- The matrix A is randomly generated for each test.
- The following scaled residual check will be computed:
      ||Ax-b||_oo / ( eps * ( || x ||_oo * || A ||_oo + || b ||_oo ) * N )
- The relative machine precision (eps) is taken to be               1.110223e-16
- Computational tests pass if scaled residuals are less than                16.0

================================================================================
T/V                N    NB     P     Q               Time                 Gflops
--------------------------------------------------------------------------------
WR11C2R4      239872   256    12    16            3071.39             2.9958e+03
HPL_pdgesv() start time Tue Oct 29 01:17:45 2024

HPL_pdgesv() end time   Tue Oct 29 02:08:56 2024

--------------------------------------------------------------------------------
||Ax-b||_oo/(eps*(||A||_oo*||x||_oo+||b||_oo)*N)=   1.56457026e-03 ...... PASSED
================================================================================

Finished      1 tests with the following results:
              1 tests completed and passed residual checks,
              0 tests completed and failed residual checks,
              0 tests skipped because of illegal input values.
--------------------------------------------------------------------------------

End of Tests.
================================================================================
```

[image]

geerlingguy, 2024-10-29:
If I attempt to modify my config to use the `configure` options in the Ampere-supplied repo, I wind up with errors compiling HPL. I've noticed another major difference is the HPL Makefile, `Make.AmpereOne` — it differs a bit from what I generate in the task that builds `sh make_generic`...

I'm happy enough accepting the result from the Ampere build though. To be more complete, I should probably denote runtime options used by each run in the issues—I don't *always* do that, but if there are platform-specific optimizations (besides `auto` and `make_generic`), that can lead to a substantial difference in performance.

geerlingguy, 2024-10-29:
Going to run once more tonight, to see if I can pass 3 Tflops (got 2.996 lol).

geerlingguy, 2024-10-29:
Also to give a little more of a leg up, I may do one more run with 25 Gbps network detached, and pulling all the NVMe drives out from the front. I'm guessing each drive sucks down 2-5W at idle...

geerlingguy, 2024-10-29:
Today's run, without NVMe, and without 25G networking connected, was rock solid at 692W (see graph below), for:

**3,027 Gflops at 692W, or 4.37 Gflops/W**

[image]

[image]

geerlingguy, 2024-10-29:
I may do another run later downclocked to 2.6 GHz or some lower clock, to see if efficiency can scale up a bit, and what the sweet spot is for the AmpereOne.

geerlingguy, 2024-10-30:
Setting the frequency to 2.6 GHz all cores to simulate the more efficient A192-26X SKU:

```
sudo cpupower frequency-set -u 2600000
```

2,755 Gflops at 612W, for 4.50 Gfops/W

```
================================================================================
HPLinpack 2.3  --  High-Performance Linpack benchmark  --   December 2, 2018
Written by A. Petitet and R. Clint Whaley,  Innovative Computing Laboratory, UTK
Modified by Piotr Luszczek, Innovative Computing Laboratory, UTK
Modified by Julien Langou, University of Colorado Denver
================================================================================

An explanation of the input/output parameters follows:
T/V    : Wall time / encoded variant.
N      : The order of the coefficient matrix A.
NB     : The partitioning blocking factor.
P      : The number of process rows.
Q      : The number of process columns.
Time   : Time in seconds to solve the linear system.
Gflops : Rate of execution for solving the linear system.

The following parameter values will be used:

N      :  239872 
NB     :     256 
PMAP   : Row-major process mapping
P      :      12 
Q      :      16 
PFACT  :   Right 
NBMIN  :       4 
NDIV   :       2 
RFACT  :   Crout 
BCAST  :  1ringM 
DEPTH  :       1 
SWAP   : Mix (threshold = 64)
L1     : transposed form
U      : transposed form
EQUIL  : yes
ALIGN  : 8 double precision words

--------------------------------------------------------------------------------

- The matrix A is randomly generated for each test.
- The following scaled residual check will be computed:
      ||Ax-b||_oo / ( eps * ( || x ||_oo * || A ||_oo + || b ||_oo ) * N )
- The relative machine precision (eps) is taken to be               1.110223e-16
- Computational tests pass if scaled residuals are less than                16.0

================================================================================
T/V                N    NB     P     Q               Time                 Gflops
--------------------------------------------------------------------------------
WR11C2R4      239872   256    12    16            3340.34             2.7546e+03
HPL_pdgesv() start time Wed Oct 30 01:52:32 2024

HPL_pdgesv() end time   Wed Oct 30 02:48:12 2024

--------------------------------------------------------------------------------
||Ax-b||_oo/(eps*(||A||_oo*||x||_oo+||b||_oo)*N)=   1.58219282e-03 ...... PASSED
================================================================================

Finished      1 tests with the following results:
              1 tests completed and passed residual checks,
              0 tests completed and failed residual checks,
              0 tests skipped because of illegal input values.
--------------------------------------------------------------------------------

End of Tests.
================================================================================
```

geerlingguy, 2024-10-30:
Closing this issue for now as I have the results—I will reopen if we get any other optimizations back from Ampere!

geerlingguy, 2024-10-30:
One more run this morning, with the 6x NVMe drives unplugged, for max efficiency score...

2,745.1 Gflops at 570 W, for 4.82 Gflops/W

```
================================================================================
HPLinpack 2.3  --  High-Performance Linpack benchmark  --   December 2, 2018
Written by A. Petitet and R. Clint Whaley,  Innovative Computing Laboratory, UTK
Modified by Piotr Luszczek, Innovative Computing Laboratory, UTK
Modified by Julien Langou, University of Colorado Denver
================================================================================

An explanation of the input/output parameters follows:
T/V    : Wall time / encoded variant.
N      : The order of the coefficient matrix A.
NB     : The partitioning blocking factor.
P      : The number of process rows.
Q      : The number of process columns.
Time   : Time in seconds to solve the linear system.
Gflops : Rate of execution for solving the linear system.

The following parameter values will be used:

N      :  239872 
NB     :     256 
PMAP   : Row-major process mapping
P      :      12 
Q      :      16 
PFACT  :   Right 
NBMIN  :       4 
NDIV   :       2 
RFACT  :   Crout 
BCAST  :  1ringM 
DEPTH  :       1 
SWAP   : Mix (threshold = 64)
L1     : transposed form
U      : transposed form
EQUIL  : yes
ALIGN  : 8 double precision words

--------------------------------------------------------------------------------

- The matrix A is randomly generated for each test.
- The following scaled residual check will be computed:
      ||Ax-b||_oo / ( eps * ( || x ||_oo * || A ||_oo + || b ||_oo ) * N )
- The relative machine precision (eps) is taken to be               1.110223e-16
- Computational tests pass if scaled residuals are less than                16.0

================================================================================
T/V                N    NB     P     Q               Time                 Gflops
--------------------------------------------------------------------------------
WR11C2R4      239872   256    12    16            3351.97             2.7451e+03
HPL_pdgesv() start time Wed Oct 30 14:23:21 2024

HPL_pdgesv() end time   Wed Oct 30 15:19:13 2024

--------------------------------------------------------------------------------
||Ax-b||_oo/(eps*(||A||_oo*||x||_oo+||b||_oo)*N)=   1.55559315e-03 ...... PASSED
================================================================================

Finished      1 tests with the following results:
              1 tests completed and passed residual checks,
              0 tests completed and failed residual checks,
              0 tests skipped because of illegal input values.
--------------------------------------------------------------------------------

End of Tests.
================================================================================
```

[image]

closed 2024-10-30 by geerlingguy

geerlingguy, 2024-10-31:
(Just to be complete, after getting the extra stick of RAM, I re-ran the benchmark and got about the same 3 Tflop result)