geerlingguy/top500-benchmark #50: Benchmark Raspberry Pi 500

geerlingguy, 2024-12-09:
As the title says—see https://github.com/geerlingguy/sbc-reviews/issues/60

geerlingguy, 2024-12-09:
29.812 Gflops at 11.5W for **2.59 Gflops/W**

```
================================================================================
HPLinpack 2.3  --  High-Performance Linpack benchmark  --   December 2, 2018
Written by A. Petitet and R. Clint Whaley,  Innovative Computing Laboratory, UTK
Modified by Piotr Luszczek, Innovative Computing Laboratory, UTK
Modified by Julien Langou, University of Colorado Denver
================================================================================

An explanation of the input/output parameters follows:
T/V    : Wall time / encoded variant.
N      : The order of the coefficient matrix A.
NB     : The partitioning blocking factor.
P      : The number of process rows.
Q      : The number of process columns.
Time   : Time in seconds to solve the linear system.
Gflops : Rate of execution for solving the linear system.

The following parameter values will be used:

N      :   23314
NB     :     256
PMAP   : Row-major process mapping
P      :       1
Q      :       4
PFACT  :   Right
NBMIN  :       4
NDIV   :       2
RFACT  :   Crout
BCAST  :  1ringM
DEPTH  :       1
SWAP   : Mix (threshold = 64)
L1     : transposed form
U      : transposed form
EQUIL  : yes
ALIGN  : 8 double precision words

--------------------------------------------------------------------------------

- The matrix A is randomly generated for each test.
- The following scaled residual check will be computed:
      ||Ax-b||_oo / ( eps * ( || x ||_oo * || A ||_oo + || b ||_oo ) * N )
- The relative machine precision (eps) is taken to be               1.110223e-16
- Computational tests pass if scaled residuals are less than                16.0

================================================================================
T/V                N    NB     P     Q               Time                 Gflops
--------------------------------------------------------------------------------
WR11C2R4       23314   256     1     4             283.41             2.9812e+01
HPL_pdgesv() start time Fri Nov 29 14:04:58 2024

HPL_pdgesv() end time   Fri Nov 29 14:09:42 2024

--------------------------------------------------------------------------------
||Ax-b||_oo/(eps*(||A||_oo*||x||_oo+||b||_oo)*N)=   3.83945609e-03 ...... PASSED
================================================================================

Finished      1 tests with the following results:
              1 tests completed and passed residual checks,
              0 tests completed and failed residual checks,
              0 tests skipped because of illegal input values.
--------------------------------------------------------------------------------

End of Tests.
================================================================================
```

The Pi 500 didn't thermal throttle:

```
pi@pi500:~/Downloads/top500-benchmark $ vcgencmd get_throttled
throttled=0x0
```

geerlingguy, 2024-12-09:
## 2.8 GHz Overclock

I attempted a 3.0 GHz Overclock, but that speed was unstable on the Pi 500; it resulted in some segfaults and a lockup after a while. But at 2.8 GHz:

```
over_voltage_delta=50000
arm_freq=2800
gpu_freq=1000
```

34.318 Gflops at 15W for **2.29 Gflops/W**

```
================================================================================
HPLinpack 2.3  --  High-Performance Linpack benchmark  --   December 2, 2018
Written by A. Petitet and R. Clint Whaley,  Innovative Computing Laboratory, UTK
Modified by Piotr Luszczek, Innovative Computing Laboratory, UTK
Modified by Julien Langou, University of Colorado Denver
================================================================================

An explanation of the input/output parameters follows:
T/V    : Wall time / encoded variant.
N      : The order of the coefficient matrix A.
NB     : The partitioning blocking factor.
P      : The number of process rows.
Q      : The number of process columns.
Time   : Time in seconds to solve the linear system.
Gflops : Rate of execution for solving the linear system.

The following parameter values will be used:

N      :   23314
NB     :     256
PMAP   : Row-major process mapping
P      :       1
Q      :       4
PFACT  :   Right
NBMIN  :       4
NDIV   :       2
RFACT  :   Crout
BCAST  :  1ringM
DEPTH  :       1
SWAP   : Mix (threshold = 64)
L1     : transposed form
U      : transposed form
EQUIL  : yes
ALIGN  : 8 double precision words

--------------------------------------------------------------------------------

- The matrix A is randomly generated for each test.
- The following scaled residual check will be computed:
      ||Ax-b||_oo / ( eps * ( || x ||_oo * || A ||_oo + || b ||_oo ) * N )
- The relative machine precision (eps) is taken to be               1.110223e-16
- Computational tests pass if scaled residuals are less than                16.0

================================================================================
T/V                N    NB     P     Q               Time                 Gflops
--------------------------------------------------------------------------------
WR11C2R4       23314   256     1     4             246.20             3.4318e+01
HPL_pdgesv() start time Mon Dec  2 12:52:13 2024

HPL_pdgesv() end time   Mon Dec  2 12:56:19 2024

--------------------------------------------------------------------------------
||Ax-b||_oo/(eps*(||A||_oo*||x||_oo+||b||_oo)*N)=   3.83945609e-03 ...... PASSED
================================================================================

Finished      1 tests with the following results:
              1 tests completed and passed residual checks,
              0 tests completed and failed residual checks,
              0 tests skipped because of illegal input values.
--------------------------------------------------------------------------------

End of Tests.
================================================================================
```

The Pi 500 thermal throttled:

```
pi@pi500:~/Downloads/top500-benchmark $ vcgencmd get_throttled
throttled=0xe0000
```

Those three bits:

```
              17    Arm frequency capping has occurred
              18    Throttling has occurred
              19    Soft temperature limit has occurred
```

geerlingguy, 2024-12-09:
Running once more with launch-day Pi OS updates, which enable NUMA/SDRAM tweaks automatically...

TODO

closed 2024-12-09 by geerlingguy

popcornmix, 2025-02-03:
I was investigating why (with some linpack settings), the scores can go down with BANKLOW=1, NUMA=8.
It turns out mpiexec (used when launching linpack) overwrites the default numa settings (effectively setting it to NUMA=1).

I've worked out that adding `-bind-to none` e.g.
```
mpiexec -bind-to none -np 4 xhpl
```
avoids that unwanted behaviour and gave me higher scores (when using NUMA enabled bootloader/kernel).

Perhaps that will help your scores.

geerlingguy, 2025-02-03:
@popcornmix - Good catch! I opened a follow-up issue https://github.com/geerlingguy/top500-benchmark/issues/55 — hopefully I can get some time to test different systems to check if this behavior affects other things besides the Pi by default as well... did you run the top500 benchmark on a Pi 500 and get a noticeable difference in scores?

popcornmix, 2025-02-04:
The difference seemed to vary with exact HPL.dat config options, but I got:
23.073 Gflops without "bind-to none" and 35.663 Gflops with "bind-to none"

This was a Pi 5 with no overclock.
It wasn't clear to me if your most recent report in this issue was with or without overclock (I see the second report was overclocked).

geerlingguy, 2025-02-04:
@popcornmix - My result with no overclock for the Pi 500 was 35.586 Gflops (see official results table here: https://github.com/geerlingguy/top500-benchmark?tab=readme-ov-file#results) — I typically put results in with all the factory defaults and latest OS release, and these issues are just for messing around with various options to see better or worse performance results.