geerlingguy/top500-benchmark #54: Benchmark Radxa Oryon O6 Mini ITX board

geerlingguy, 2025-01-31:
See: https://github.com/geerlingguy/sbc-reviews/issues/62

With Ps = 3 and Qs = 4, I got a fairly abysmal 36.014 Gflops at 35.7 W for **1.01 Glops/W**

Definitely more work needed to get full speed out of this chip with an HPL run. I think I'll have to try to stick to the performance cores (A720), maybe, and ditch the slower A520s.

```
================================================================================
HPLinpack 2.3  --  High-Performance Linpack benchmark  --   December 2, 2018
Written by A. Petitet and R. Clint Whaley,  Innovative Computing Laboratory, UTK
Modified by Piotr Luszczek, Innovative Computing Laboratory, UTK
Modified by Julien Langou, University of Colorado Denver
================================================================================

An explanation of the input/output parameters follows:
T/V    : Wall time / encoded variant.
N      : The order of the coefficient matrix A.
NB     : The partitioning blocking factor.
P      : The number of process rows.
Q      : The number of process columns.
Time   : Time in seconds to solve the linear system.
Gflops : Rate of execution for solving the linear system.

The following parameter values will be used:

N      :   34581
NB     :     256
PMAP   : Row-major process mapping
P      :       3
Q      :       4
PFACT  :   Right
NBMIN  :       4
NDIV   :       2
RFACT  :   Crout
BCAST  :  1ringM
DEPTH  :       1
SWAP   : Mix (threshold = 64)
L1     : transposed form
U      : transposed form
EQUIL  : yes
ALIGN  : 8 double precision words

--------------------------------------------------------------------------------

- The matrix A is randomly generated for each test.
- The following scaled residual check will be computed:
      ||Ax-b||_oo / ( eps * ( || x ||_oo * || A ||_oo + || b ||_oo ) * N )
- The relative machine precision (eps) is taken to be               1.110223e-16
- Computational tests pass if scaled residuals are less than                16.0

================================================================================
T/V                N    NB     P     Q               Time                 Gflops
--------------------------------------------------------------------------------
WR11C2R4       34581   256     3     4             765.56             3.6014e+01
HPL_pdgesv() start time Sat Feb  1 06:45:53 2025

HPL_pdgesv() end time   Sat Feb  1 06:58:39 2025

--------------------------------------------------------------------------------
||Ax-b||_oo/(eps*(||A||_oo*||x||_oo+||b||_oo)*N)=   3.41160418e-03 ...... PASSED
================================================================================

Finished      1 tests with the following results:
              1 tests completed and passed residual checks,
              0 tests completed and failed residual checks,
              0 tests skipped because of illegal input values.
--------------------------------------------------------------------------------

End of Tests.
================================================================================
```

[image]

geerlingguy, 2025-01-31:
Second run averaging more like 33.5W with Ps at 2 and Qs at 4 (trying to only target the faster A720 cores).

31.232 Gflops at 33.5 W for **0.93 Gflops/W**

```
================================================================================
HPLinpack 2.3  --  High-Performance Linpack benchmark  --   December 2, 2018
Written by A. Petitet and R. Clint Whaley,  Innovative Computing Laboratory, UTK
Modified by Piotr Luszczek, Innovative Computing Laboratory, UTK
Modified by Julien Langou, University of Colorado Denver
================================================================================

An explanation of the input/output parameters follows:
T/V    : Wall time / encoded variant.
N      : The order of the coefficient matrix A.
NB     : The partitioning blocking factor.
P      : The number of process rows.
Q      : The number of process columns.
Time   : Time in seconds to solve the linear system.
Gflops : Rate of execution for solving the linear system.

The following parameter values will be used:

N      :   34581
NB     :     256
PMAP   : Row-major process mapping
P      :       2
Q      :       4
PFACT  :   Right
NBMIN  :       4
NDIV   :       2
RFACT  :   Crout
BCAST  :  1ringM
DEPTH  :       1
SWAP   : Mix (threshold = 64)
L1     : transposed form
U      : transposed form
EQUIL  : yes
ALIGN  : 8 double precision words

--------------------------------------------------------------------------------

- The matrix A is randomly generated for each test.
- The following scaled residual check will be computed:
      ||Ax-b||_oo / ( eps * ( || x ||_oo * || A ||_oo + || b ||_oo ) * N )
- The relative machine precision (eps) is taken to be               1.110223e-16
- Computational tests pass if scaled residuals are less than                16.0

================================================================================
T/V                N    NB     P     Q               Time                 Gflops
--------------------------------------------------------------------------------
WR11C2R4       34581   256     2     4             882.78             3.1232e+01
HPL_pdgesv() start time Sat Feb  1 07:04:48 2025

HPL_pdgesv() end time   Sat Feb  1 07:19:31 2025

--------------------------------------------------------------------------------
||Ax-b||_oo/(eps*(||A||_oo*||x||_oo+||b||_oo)*N)=   3.41160418e-03 ...... PASSED
================================================================================

Finished      1 tests with the following results:
              1 tests completed and passed residual checks,
              0 tests completed and failed residual checks,
              0 tests skipped because of illegal input values.
--------------------------------------------------------------------------------

End of Tests.
================================================================================
```

geerlingguy, 2025-03-26:
Going to close this for now—if newer firmware is released that fixes the core cluster scheduling issues, idle power draw, and clocks/memory bandwidth, I'll re-open and test again.

geerlingguy, 2025-04-27:
I just re-ran HPL today with BIOS version 9.0.0 and Ubuntu 25.04... and it's not wonderful:

20.161 Gflops at 26.4W for **0.76 Gflops/W**

```
================================================================================
HPLinpack 2.3  --  High-Performance Linpack benchmark  --   December 2, 2018
Written by A. Petitet and R. Clint Whaley,  Innovative Computing Laboratory, UTK
Modified by Piotr Luszczek, Innovative Computing Laboratory, UTK
Modified by Julien Langou, University of Colorado Denver
================================================================================

An explanation of the input/output parameters follows:
T/V    : Wall time / encoded variant.
N      : The order of the coefficient matrix A.
NB     : The partitioning blocking factor.
P      : The number of process rows.
Q      : The number of process columns.
Time   : Time in seconds to solve the linear system.
Gflops : Rate of execution for solving the linear system.

The following parameter values will be used:

N      :   34581
NB     :     256
PMAP   : Row-major process mapping
P      :       1
Q      :       4
PFACT  :   Right
NBMIN  :       4
NDIV   :       2
RFACT  :   Crout
BCAST  :  1ringM
DEPTH  :       1
SWAP   : Mix (threshold = 64)
L1     : transposed form
U      : transposed form
EQUIL  : yes
ALIGN  : 8 double precision words

--------------------------------------------------------------------------------

- The matrix A is randomly generated for each test.
- The following scaled residual check will be computed:
      ||Ax-b||_oo / ( eps * ( || x ||_oo * || A ||_oo + || b ||_oo ) * N )
- The relative machine precision (eps) is taken to be               1.110223e-16
- Computational tests pass if scaled residuals are less than                16.0

================================================================================
T/V                N    NB     P     Q               Time                 Gflops
--------------------------------------------------------------------------------
WR11C2R4       34581   256     1     4            1367.50             2.0161e+01
HPL_pdgesv() start time Sat Apr 26 21:27:08 2025

HPL_pdgesv() end time   Sat Apr 26 21:49:56 2025

--------------------------------------------------------------------------------
||Ax-b||_oo/(eps*(||A||_oo*||x||_oo+||b||_oo)*N)=   3.41160418e-03 ...... PASSED
================================================================================

Finished      1 tests with the following results:
              1 tests completed and passed residual checks,
              0 tests completed and failed residual checks,
              0 tests skipped because of illegal input values.
--------------------------------------------------------------------------------

End of Tests.
================================================================================
```

volyrique, 2025-05-08:
@geerlingguy, my bad, `arm64` turned out to be a red herring - I was probably experimenting with another option at the same time, which was why I noticed an improvement. Unless I have made another mistake, with your current Ansible setup you should just use `generic`.

However, the highest performance that I have achieved so far is by doing a manual build in the following manner: I use the distro-provided `mpich` package (I doubt that it has that much of an effect on performance, but I might try recompiling it myself at some point). I build BLIS with the `arm64` configuration, but I remove `armsve` from the line starting with `arm64:` inside the `config_registry` file at the top of the source tree before that. In the makefile used by HPL I set `HPL_OPTS` to `-flto -march=armv9.2-a+nosve -mtune=native -O3` and `LINKFLAGS` to `$(CCFLAGS)`. The rest is the same as the Ansible setup; my board has 32 GiB RAM BTW. With that I run `mpirun -np 12 ./xhpl` and get:
```
================================================================================
HPLinpack 2.3  --  High-Performance Linpack benchmark  --   December 2, 2018
Written by A. Petitet and R. Clint Whaley,  Innovative Computing Laboratory, UTK
Modified by Piotr Luszczek, Innovative Computing Laboratory, UTK
Modified by Julien Langou, University of Colorado Denver
================================================================================

An explanation of the input/output parameters follows:
T/V    : Wall time / encoded variant.
N      : The order of the coefficient matrix A.
NB     : The partitioning blocking factor.
P      : The number of process rows.
Q      : The number of process columns.
Time   : Time in seconds to solve the linear system.
Gflops : Rate of execution for solving the linear system.

The following parameter values will be used:

N      :   48905 
NB     :     256 
PMAP   : Row-major process mapping
P      :       1 
Q      :      12 
PFACT  :   Right 
NBMIN  :       4 
NDIV   :       2 
RFACT  :   Crout 
BCAST  :  1ringM 
DEPTH  :       1 
SWAP   : Mix (threshold = 64)
L1     : transposed form
U      : transposed form
EQUIL  : yes
ALIGN  : 8 double precision words

--------------------------------------------------------------------------------

- The matrix A is randomly generated for each test.
- The following scaled residual check will be computed:
      ||Ax-b||_oo / ( eps * ( || x ||_oo * || A ||_oo + || b ||_oo ) * N )
- The relative machine precision (eps) is taken to be               1.110223e-16
- Computational tests pass if scaled residuals are less than                16.0

================================================================================
T/V                N    NB     P     Q               Time                 Gflops
--------------------------------------------------------------------------------
WR11C2R4       48905   256     1    12             840.47             9.2782e+01
HPL_pdgesv() start time Thu May  8 07:06:03 2025

HPL_pdgesv() end time   Thu May  8 07:20:03 2025

--------------------------------------------------------------------------------
||Ax-b||_oo/(eps*(||A||_oo*||x||_oo+||b||_oo)*N)=   2.36019821e-03 ...... PASSED
================================================================================

Finished      1 tests with the following results:
              1 tests completed and passed residual checks,
              0 tests completed and failed residual checks,
              0 tests skipped because of illegal input values.
--------------------------------------------------------------------------------

End of Tests.
================================================================================
```
I wonder if I could break 100 Gflops, e.g. by using `-flto -march=armv9.2-a+nosve -mtune=native` while compiling BLIS, but the build system is a bit obtuse for me 😄.

volyrique, 2025-09-17:
FYI pull request #70 contains the majority of the changes that I mentioned in my previous message. I also suggested setting `blis_configure_options` to `generic`, but `cortexa57` seems to work better; in fact, it is equivalent to the `armsve` hack that I used.

geerlingguy, 2025-11-11:
- Defaults, Ps = 1 and Qs = 12: 36.90 Gflops at 36W, for **1.025 Gflops/W**
  - Manual blis compilation, `auto` selecting `armsve`, Ps = 1 and Qs = 12: 41.20 Gflops at 36W, for **1.144 Gflops/W**
  - Manual blis compilation, forcing `cortexa57`, Ps = 1 and Qs = 12: TODO Gflops at TODOW, for **TODO Gflops/W**

Same as with https://github.com/geerlingguy/top500-benchmark/issues/82#issuecomment-3517793190, when I manually compiled blis with `sudo ./configure --prefix="/opt/top500/blis" auto`, I got:

```
configure: automatic configuration requested.
configure: hardware detection driver returned 'armsve'.
configure: checking configuration against contents of 'config_registry'.
configure: configuration 'armsve' is registered.
configure: 'armsve' is defined as having the following sub-configurations:
configure:    armsve
configure: which collectively require the following kernels:
configure:    armsve
configure: checking sub-configurations:
configure:   'armsve' is registered...and exists.
configure: checking sub-configurations' requisite kernels:
configure:   'armsve' kernels...exist.
```

geerlingguy, 2025-11-11:
@volyrique - I understand — what I'm doing though, is testing every possibility to try to see what's happening.

On the MS-R1, it looks like `armsve` is giving a slightly better benchmark than `cortexa57` with the latest blis, but both are very close. I'm re-testing every assumption and possibility, because the only thing I know about the Cix SoC is it behaves strange, and also seems to be somewhat environment-dependent (with multiple firmwares and OS support levels depending on your specific setup).

More so than other things I've tested.

I'm back on the MS-R1 now though, because my re-test is going back to the older Gflops levels, and I'm trying to figure out what change I made (or maybe it was the phase of a moon yesterday?) that made it go from 40-ish Gflops to 140 :D

**Separately**: Where are you getting newer BIOS releases? I'm scouring the [Orion O6 BIOS](https://docs.radxa.com/en/orion/o6/bios/install-bios) docs and can't find anything newer than the SystemReady 9.0.0 or the 0.3.0-1 vendor BIOS: https://docs.radxa.com/en/orion/o6/download#bios

segabor, 2025-11-11:
@geerlingguy AFAIK option 28 was the right one

volyrique, 2025-11-11:
@geerlingguy Glad that you got this sorted. I guess that you don't need any more input from me now? BTW you are kind of safe with respect to this issue for the time being, as I said before, but as newer SoCs enter the market, that may no longer be the case. For instance, Rockchip's successor to the RK3588, [RK3688](https://www.cnx-software.com/2024/10/04/rockchip-rk3688-armv9-3-aiot-processor-to-feature-a-16-tops-npu-ufs-4-0-storage), most certainly does support SVE with 128-bit vectors and, who knows, maybe Raspberry Pi 6 will too?

The only crashes that I am experiencing right now are either thermal- or GPU-related, I believe. I was trying to get another set of HPL results, but my machine kept shutting itself down, necessitating a hard reset (which was why it was taking me so long); FYI I am on an 850 W ATX power supply. The GPU issues manifest themselves after I start running llama.cpp repeatedly and they might also be thermal-related, to be fair, just affecting the graphics card instead of the CPU.

geerlingguy, 2025-11-12:
> Rockchip's successor to the RK3588, [RK3688](https://www.cnx-software.com/2024/10/04/rockchip-rk3688-armv9-3-aiot-processor-to-feature-a-16-tops-npu-ufs-4-0-storage), most certainly does support SVE with 128-bit vectors and, who knows, maybe Raspberry Pi 6 will too?

Yeah, I've left a note in the `example.config.yml` now to remind me in case I forget, so I don't have to go digging for this issue again.

I've also updated my blog post (https://www.jeffgeerling.com/blog/2025/minisforum-stuffs-entire-arm-homelab-ms-r1) with a few notes about this issue (and a mention of your suggestion!).

> The GPU issues manifest themselves after I start running llama.cpp repeatedly and they might also be thermal-related, to be fair, just affecting the graphics card instead of the CPU.

Yeah, in this case I just have no clue, no rhyme or reason. A couple times it locked up before I started doing anything at all. Other times it was right when compilation would hit all the cores. Really seems either thermal or power, but the last couple hours of testing today it was stable, so who knows...

Edit: Also, thanks so much for the help!