geerlingguy/top500-benchmark #57: Benchmark M4 Max Mac Studio

geerlingguy, 2025-03-12:
I have one now...

```
  Model Name:	Mac Studio
  Model Identifier:	Mac16,9
  Model Number:	Z1CD0017ZLL/A
  Chip:	Apple M4 Max
  Total Number of Cores:	16 (12 performance and 4 efficiency)
  Memory:	64 GB
```

geerlingguy, 2025-03-12:
Power monitoring:

  - Idle, with monitor, USB keyboard, and mouse (no Ethernet): 7.1W
  - Idle, with all the above plus 10 Gbps Ethernet: 12W
  - Under load, top500 benchmark running: TODO

geerlingguy, 2025-03-12:
Running the benchmark:

  - The SoC temp went up to around 95°C, with the fans barely moving air, for the first couple minutes
  - The power (at the wall) went up to around 119W
  - Power remained stable, as the temperature climbed slowly up to 105°C
  - At that point, the fans ramped up to an audible level, but it was pretty decent white noise, the fan must be rather large.
  - The air coming out the back was not that hot.

Another interesting note: the total power as reported by the Mac in iStat Menus seemed to match very closely to the wall power measurement—nice!

[image]

geerlingguy, 2025-03-12:
And with 4/4:

685.00 Gflops at 120W for **5.71 Gflops/W**

```
================================================================================
HPLinpack 2.3  --  High-Performance Linpack benchmark  --   December 2, 2018
Written by A. Petitet and R. Clint Whaley,  Innovative Computing Laboratory, UTK
Modified by Piotr Luszczek, Innovative Computing Laboratory, UTK
Modified by Julien Langou, University of Colorado Denver
================================================================================

An explanation of the input/output parameters follows:
T/V    : Wall time / encoded variant.
N      : The order of the coefficient matrix A.
NB     : The partitioning blocking factor.
P      : The number of process rows.
Q      : The number of process columns.
Time   : Time in seconds to solve the linear system.
Gflops : Rate of execution for solving the linear system.

The following parameter values will be used:

N      :   71481
NB     :     256
PMAP   : Row-major process mapping
P      :       4
Q      :       4
PFACT  :   Right
NBMIN  :       4
NDIV   :       2
RFACT  :   Crout
BCAST  :  1ringM
DEPTH  :       1
SWAP   : Mix (threshold = 64)
L1     : transposed form
U      : transposed form
EQUIL  : yes
ALIGN  : 8 double precision words

--------------------------------------------------------------------------------

- The matrix A is randomly generated for each test.
- The following scaled residual check will be computed:
      ||Ax-b||_oo / ( eps * ( || x ||_oo * || A ||_oo + || b ||_oo ) * N )
- The relative machine precision (eps) is taken to be               1.110223e-16
- Computational tests pass if scaled residuals are less than                16.0

================================================================================
T/V                N    NB     P     Q               Time                 Gflops
--------------------------------------------------------------------------------
WR11C2R4       71481   256     4     4             355.47             6.8500e+02
HPL_pdgesv() start time Wed Mar 12 21:16:49 2025

HPL_pdgesv() end time   Wed Mar 12 21:22:45 2025

--------------------------------------------------------------------------------
||Ax-b||_oo/(eps*(||A||_oo*||x||_oo+||b||_oo)*N)=   3.19645745e-03 ...... PASSED
================================================================================

Finished      1 tests with the following results:
              1 tests completed and passed residual checks,
              0 tests completed and failed residual checks,
              0 tests skipped because of illegal input values.
--------------------------------------------------------------------------------

End of Tests.
================================================================================
```

geerlingguy, 2025-03-12:
Docker Desktop settings, FYI:

[image]

geerlingguy, 2025-03-12:
Geekbench 6 result: 3896 single core, 25660 multi core - https://browser.geekbench.com/v6/cpu/10992635

Though I think Geekbench could get a _lot_ better multicore score, it's been known for a while Geekbench 6 scales poorly for more than a few cores. The ratio here is only 6.6x or so. On Geekbench 5 it's much higher.

Just looking at power draw, look at how the HPL tests pull power versus Geekbench (the last spiky bits):

[image]

geerlingguy, 2025-03-12:
Quick NVMe SSD speed tests with built-in storage (I have the 1TB option):

[image]

It had spiked up to about 5 GB/sec but not beyond that.

closed 2025-03-13 by geerlingguy

i0ntempest, 2025-03-14:
Curious what linpack did you link the hpl binary to? I have the same machine (16C/64G), hpl installed from macports and linked Apple Accelerate framework. Using the same benchmark settings (4/4 Ps/Qs) I got 841.4 Gflops at 95W reported by iStat Menus. I also tried linking to OpenBLAS but that run took too long that I aborted it.

[image]

geerlingguy, 2025-03-15:
I just used the default config (https://github.com/geerlingguy/top500-benchmark/blob/master/example.config.yml) with the Ps/Qs settings I mentioned above — running within the Docker container (which introduces a little overhead from the Docker VM).

See: https://github.com/geerlingguy/top500-benchmark?tab=readme-ov-file#benchmarking---single-node

I'd love to make it easier to run this natively on macOS as well.

i0ntempest, 2025-03-15:
So you're running the benchmark under virtualization, that makes sense.
With MacPorts you can install hpl natively with `sudo port install hpl +accelerate` (or `+atlas`, `+openblas`, `+blis` if you want). Could be something worth adding into your playbooks.

geerlingguy, 2025-03-15:
@i0ntempest moved that over to a new issue so I can work on getting it automated https://github.com/geerlingguy/top500-benchmark/issues/58