geerlingguy/top500-benchmark #58: HPL - run natively on macOS

geerlingguy, 2025-03-15:
According to @i0ntempest's testing, there's probably a 5-10% advantage gained by not running the HPL benchmarks inside Docker containers on macOS.

Right now I do that just because the setup process I have dialed in for Linux doesn't run on macOS natively, at least not super-easily...

With MacPorts: `sudo port install hpl +accelerate` (or `+atlas`, `+openblas`, `+blis`), need to see if there's a simple way to get it installed (or compile it properly) on macOS without MacPorts set up.

i0ntempest, 2025-03-16:
Well if you're not using MacPorts your only option would be manual compilation (I don't think homebrew has `hpl`). Pulling and compiling all the libraries and compilers and (crucially) compile them in the right configuration for maximum performance can be pretty complicated.

geerlingguy, 2025-03-18:
@i0ntempest - Though that's exactly what I do on the Linux side (thus this repsitory's existence—to make that process automated and repeatable across a variety of arch's!).