geerlingguy/top500-benchmark #72: Benchmark for single-socket EPYC 9755

anematode, 2025-09-19:
Maybe interesting to you

```
An explanation of the input/output parameters follows:
T/V    : Wall time / encoded variant.
N      : The order of the coefficient matrix A.
NB     : The partitioning blocking factor.
P      : The number of process rows.
Q      : The number of process columns.
Time   : Time in seconds to solve the linear system.
Gflops : Rate of execution for solving the linear system.

The following parameter values will be used:

N      :  368904
NB     :     456
PMAP   : Row-major process mapping
P      :       1
Q      :       1
PFACT  :   Crout
NBMIN  :      38
NDIV   :       3
RFACT  :   Right
BCAST  :  IBcast
DEPTH  :       0                                                                                                                 
SWAP   : (211) Spread-roll                                                                                                       
L1     : transposed form                                                                                                         
A      : no-transposed form (ColMajor)                                                                                           
EQUIL  : no                                                                                                                      
ALIGN  : 8 double precision words
MXSWP  : (1/c) Collective 

--------------------------------------------------------------------------------

- The matrix A is randomly generated for each test.
- The following scaled residual check will be computed:
      ||Ax-b||_oo / ( eps * ( || x ||_oo * || A ||_oo + || b ||_oo ) * N )
- The relative machine precision (eps) is taken to be               1.110223e-16
- Computational tests pass if scaled residuals are less than                16.0

================================================================================
T/V         SWP           N    NB     P     Q            Time             Gflops
--------------------------------------------------------------------------------
WRC06R3C38c 211       368904   456     1     1         5777.05         5.7936e+03
HPL_pdgesv() start time Fri Sep 19 13:49:11 2025
HPL_pdgesv() end time   Fri Sep 19 15:25:28 2025
--------------------------------------------------------------------------------
||Ax-b||_oo/(eps*(||A||_oo*||x||_oo+||b||_oo)*N)=   4.89993517e-03 ...... PASSED
================================================================================

Finished      1 tests with the following results:
              1 tests completed and passed residual checks,
              0 tests completed and failed residual checks,
              0 tests skipped because of illegal input values.
--------------------------------------------------------------------------------

End of Tests.
================================================================================
```

The measured power draw from the wall is pretty stable at 800W

geerlingguy, 2025-09-20:
Wow, 5.8 Tflops! A few questions:

  - Did you run this from this repository's playbook, or with your own tuned installation?
  - Have you tried any other configurations for Ps/Qs? It seems with different NUMA layouts, you might get better (or worse) performance?
  - Is this running in a cloud environment, or on a system in your facility (e.g. Gigabyte or whatever rackmount server?)

geerlingguy, 2025-09-20:
(Adding planned label as I'd like to add it to the result data later.)

anematode, 2025-09-20:
Have you tried benchmarking on Graviton 4, btw? I'd expect its SVE implementation to be very good

anematode, 2025-09-20:
Ok, I tried using this repository, using blis under the _skx_ configuration (they don't have a zen4 or zen5 configuration; without that, it was going down a generic code path) and with P=8, Q=16:

```
================================================================================                                             20:26:40 [29/1918]
HPLinpack 2.3  --  High-Performance Linpack benchmark  --   December 2, 2018                                                             
Written by A. Petitet and R. Clint Whaley,  Innovative Computing Laboratory, UTK                                                         
Modified by Piotr Luszczek, Innovative Computing Laboratory, UTK                                                                     
Modified by Julien Langou, University of Colorado Denver                                                                             
================================================================================                                                         
                                                                                                                                     
An explanation of the input/output parameters follows:                                                                               
T/V    : Wall time / encoded variant.                                                                                                
N      : The order of the coefficient matrix A.                                                                                      
NB     : The partitioning blocking factor.                                                                                           
P      : The number of process rows.                                                                                                 
Q      : The number of process columns.                                                                                              
Time   : Time in seconds to solve the linear system.                                                                                 
Gflops : Rate of execution for solving the linear system.                                                                            
                                                                                                                                     
The following parameter values will be used:                                                                                         
                                                                                                                                     
N      :  368904                                                                                                                     
NB     :     456                                                                                                                     
PMAP   : Row-major process mapping                                                                                                   
P      :       8                                                                                                                     
Q      :      16                                                                                                                     
PFACT  :   Crout                                                                                                                     
NBMIN  :      38                                                                                                                     
NDIV   :       3                                                                                                                     
RFACT  :   Right                                                                                                                     
BCAST  :  1ringM                                                                                                                     
DEPTH  :       0                                                                                                                     
SWAP   : Spread-roll (long)                                                                                                          
L1     : transposed form                                          
U      : transposed form                                          
EQUIL  : no                                                       
ALIGN  : 8 double precision words   

--------------------------------------------------------------------------------                                                         

- The matrix A is randomly generated for each test.                    
- The following scaled residual check will be computed:                
      ||Ax-b||_oo / ( eps * ( || x ||_oo * || A ||_oo + || b ||_oo ) * N )                                                                     
- The relative machine precision (eps) is taken to be               1.110223e-16                                                               
- Computational tests pass if scaled residuals are less than                16.0                                                               

================================================================================                                                               
T/V                N    NB     P     Q               Time                 Gflops                                                               
[... cut ...]
--------------------------------------------------------------------------------                                                               

End of Tests.                      
================================================================================  
```

So almost twice as slow :( I examined the AMD-provided sample more closely and it looks to be using AMD's fork of blis, spending the vast majority of its time in https://github.com/amd/blis/blob/master/kernels/zen4/3/bli_dgemm_avx512_asm_8x24.c . I'm guessing that's the biggest cause of the performance difference, but I'm not really sure.

geerlingguy, 2025-09-21:
Interesting, I wonder if their AVX512 optimizations could hit upstream BLIS? That seems to be the same issue I had with Ampere—they made some tweaks to help their NEON implementation speed up the runs, but it was all in an Oracle fork of OpenBLAS (IIRC)... and a bit hard to decipher exactly what was changed.

I'm tempted to put the number from this repo's setup as the 'official' result in the README, but I used the optimized run for Ampere's servers, so I guess we could just use the optimized result for AMD too.

(Could also be useful to modify the Ansible playbooks to allow for easier switching to the vendor-supplied libraries, like for AMD's BLIS fork in this case).

anematode, 2025-09-21:
Yeah, I'm a little confused. My understanding is that on a single node, HPL figures should be close to the theoretical maximum FLOPs/second since it's mostly just spamming FMAs in level-3 BLAS rather than stressing the memory subsystem. For your benchmarks on Apple, did you try using their Accelerate framework?

geerlingguy, 2025-09-22:
@anematode - No, for Macs, I am running in a VM, so missing some optimizations. See https://github.com/geerlingguy/top500-benchmark/issues/58

Djip007, 2026-01-08:
> Wow, 5.8 Tflops! A few questions:

For me it is slow for the CPU the pick theory is 11059,2 ...  look we can have more tuning using MPI may be 1 per cliplet (ie par L3 cache bloc) but did not know the topology